Genesis-Embodied-AI/genesis-world #1663: [Bug]: Shadow rendering issues when the drone is at high altitude

XXLiu-HNU, 2025-09-01:
### Bug Description

I am testing `fly_route.py` in the example, and I found that when the drone flies higher, the drone's shadow becomes a cube.

### Steps to Reproduce

I'm running the official tutorial.：
https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/drone/fly_route.py

Here is the test process, please pay attention to the drone shadow：

https://github.com/user-attachments/assets/4012a719-7068-41e5-8083-ad9ed8030fe1

### Expected Behavior

Drone's shadows are rendered correctly

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 20.04
- GPU/CPU RTX 4060
- GPU-driver version 570.181
- CUDA / CUDA-toolkit version 570.181

### Release version or Commit ID

bdea10c4f5c504652ceb93e01b40acc823b6d0ce

XXLiu-HNU, 2025-09-01:
I also found that if I turned on the world frame, the shadows displayed normally 

https://github.com/user-attachments/assets/c69801f9-4c1a-45a0-a25d-daa2b914535c

duburcqa, 2025-09-01:
I found out what is the issue. I will try to find a proper way to fix it.

closed 2025-09-01 by duburcqa