Genesis-Embodied-AI/genesis-world #1727: [Bug]: Abnormal behavior occurs when training robots on heightfield terrains

lupinjia, 2025-09-16:
### Bug Description

When training go2 robot on heightfeld terrains using RL, the collision between the terrain and the robot cause some abnormal behaviors

### Steps to Reproduce

Install [genesis_lr](https://github.com/lupinjia/genesis_lr) with an env of python=3.10, run the training session using:
```bash
python train.py --task=go2_ts --headless
```
play it: 
```bash
python play_ts.py --task=go2_ts
```
To compare with IsaacGym, you need to create a conda env with python=3.8, and run the train and play command as above.

### Expected Behavior

Stable locomotion on heightfield terrains

### Screenshots/Videos

Training results in IsaacGym:

https://github.com/user-attachments/assets/1f40fcef-8520-4cdf-9d16-5483dca6d5e0

Training results in Genesis:

https://github.com/user-attachments/assets/28517989-eb9c-4f5e-a1ac-a297a69013e3

### Relevant log output

```shell

```

### Environment

- OS: Ubuntu 24.04
- GPU/CPU Intel i9-13900K + RTX 4090 24G
- GPU-driver version 535
- CUDA / CUDA-toolkit version 12.4

### Release version or Commit ID

version number: 0.3.3

ax-anoop, 2025-09-17:
Awesome test ! 

I've experienced something similar with G1 locomotion, where terrain has super weird behaviour.

jgillick, 2025-09-19:
I'm experiencing the same issue.

https://github.com/user-attachments/assets/5c54c7b0-52ea-491e-a550-9ae566ff1f58

This is right after reset, the robot is placed slightly above the terrain and when it lands it's launched, which causes a reset.

It almost looks like the collision causes a ricochet at a much higher velocity than expected.

YilingQiao, 2025-09-22:
https://github.com/user-attachments/assets/f117845b-6157-4f48-a3f0-b78e879d3013

Can any one provide a minimal example script and asset? This code fine on my side.

```
import argparse
import time

import numpy as np
import torch

import genesis as gs

def main():

    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=True)
    parser.add_argument("-c", "--cpu", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(seed=0, backend=gs.cpu if args.cpu else gs.gpu)

    ########################## create a scene ##########################

    scene = gs.Scene(
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(-5.0, -5.0, 10.0),
            camera_lookat=(5.0, 5.0, 0.0),
            camera_fov=40,
            enable_interaction=True,
        ),
        show_viewer=args.vis,
        rigid_options=gs.options.RigidOptions(
            dt=0.01,
            constraint_solver=gs.constraint_solver.Newton,
        ),
        vis_options=gs.options.VisOptions(
            # geom_type='sdf',
        ),
    )

    horizontal_scale = 0.25
    vertical_scale = 0.005
    height_field = np.zeros([40, 40])
    heights_range = np.arange(-10, 20, 10)
    height_field[5:35, 5:35] = 200 + np.random.choice(heights_range, (30, 30))
    ########################## entities ##########################
    terrain = scene.add_entity(
        morph=gs.morphs.Terrain(
            horizontal_scale=horizontal_scale,
            vertical_scale=vertical_scale,
            height_field=height_field,
            name="example",
            # from_stored=True,
        ),
    )

    # 81e91e6c99cc883b8a57e058b56fdc080e67a2d6 Sat Jul 26 13:19:48 2025 -0700

    go2 = scene.add_entity(
        morph=gs.morphs.URDF(
            file="urdf/go2/urdf/go2.urdf",
            pos=(3.0, 3.0, 2.0),
        ),
    )
    ########################## build ##########################
    scene.build()

    height_field = terrain.geoms[0].metadata["height_field"]
    rows = horizontal_scale * torch.arange(0, height_field.shape[0], 1, device=gs.device).unsqueeze(1).repeat(
        1, height_field.shape[1]
    ).unsqueeze(-1)
    cols = horizontal_scale * torch.arange(0, height_field.shape[1], 1, device=gs.device).unsqueeze(0).repeat(
        height_field.shape[0], 1
    ).unsqueeze(-1)
    heights = vertical_scale * torch.tensor(height_field, device=gs.device).unsqueeze(-1)

    poss = torch.cat([rows, cols, heights], dim=-1).reshape(-1, 3)
    scene.draw_debug_spheres(poss=poss, radius=0.05, color=(0, 0, 1, 0.7))
    for _ in range(10000):
        # time.sleep(0.5)
        scene.step()

if __name__ == "__main__":
    main()
```

YilingQiao, 2025-09-24:
@jgillick got some bug running your repo https://github.com/jgillick/genesis-forge/issues/1

YilingQiao, 2025-09-24:
@lupinjia where to run
```
python train.py --task=go2_ts --headless
```

I did not see train.py in your repo...

jgillick, 2025-09-24:
@YilingQiao sorry about that, but thank you for reporting the bug. It should be fixed on the main branch of that project now.

YilingQiao, 2025-09-24:
@jgillick can you give me a minimal example (i.e. a script with only genesis, one single env, and a recorded control sequence that can reproduce your problematic trajectory)? I tried your repo but it's hard to extract that. You are more familiar with your repo and can do that better than me.

jgillick, 2025-09-25:
@YilingQiao I've recorded about 14 episodes of data and put a minimal example in my repo in branch `gs-bug-1727`:
https://github.com/jgillick/genesis-forge/tree/gs-bug-1727/examples/bug_1727

Pull down the entire `bug_1727` directory (it shouldn't depend on genesis forge) and run the `test.py` script. 

Not all episodes result in explosions, but something interesting happens if the episode is allowed to continue running with no further joint actions, the robot will start to bounce across the terrain.

https://github.com/user-attachments/assets/e3beec4e-0976-4252-b721-e65c3f6007b9

To see this, start by uncomment lines 141-143, then run run episode `episode_020.json` with the following command:

```shell
python ./test.py -v ./episodes/episode_020.json
```
I hope this helps

YilingQiao, 2025-09-26:
Thank you very much! The minimal example helps a lot. #1777 will fix this.

YilingQiao, 2025-09-26:
https://github.com/user-attachments/assets/fac91485-59c8-4cce-98b6-07586790b9f6

PR #1777 (2025-09-26): [BUG FIX] Fix terrain with non-zero offset

## Description

Fix #1727

The center of the object was wrong when the offset of a terrain is not zero.

## Related Issue
closed 2025-09-26 by duburcqa