Genesis-Embodied-AI/genesis-world #1740: [Bug]: Memory issue

ax-anoop, 2025-09-17:
### Bug Description

With 10K envs only 7GB of GPU is used, with 20K envs we get a mem / cuda error. 

This does not happen on older genesis version (0.2). 

[image]

### Steps to Reproduce

If possible, provide a script triggering the bug, e.g.
```python
# code snippet triggering the bug
import genesis as gs
gs.init()
scene = gs.Scene(show_viewer=False)
broken = scene.add_entity(
    gs.morphs.MJCF(file='attachement.xml'),
)
scene.build()
```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

Should run 

### Relevant log output

```shell

```

### Environment

- OS: [e.g. Ubuntu 24.04, Windows 11 24H2]
- GPU/CPU [e.g. A100, RTX 4090, M3pr, Intel I9-9900k, Ryzen 5900x] (N/A if no GPU/CPU)
- GPU-driver version (N/A if no GPU)
- CUDA / CUDA-toolkit version (N/A if non-Nvidia)

### Release version or Commit ID

-

ax-anoop, 2025-09-17:
I can't seem to pass 10K envs, though more then 70% of GPU is still available

YilingQiao, 2025-09-17:
what are you simulating? any standalone script that's easy to reproduce this?

YilingQiao, 2025-09-17:
@Kashu7100 ran go2_train example, the memoroy seems linear
```
B:4096     3578MiB /  10240MiB
B:10000   5833MiB /  10240MiB
B:20000   9787MiB /  10240MiB
```
Can you share a minimal script so we can know which function that is not covered in go2 example went wrong

ax-anoop, 2025-09-18:
**clone** > https://github.com/unitreerobotics/unitree_rl_gym 

'''
import argparse
import genesis as gs

def _select_model_path(robot: str) -> str:
    if robot == "g1_12dof":
        return "unitree_rl_gym/resources/robots/g1_description/g1_12dof.xml"
    else:
        raise ValueError(
            f"Invalid robot type: {robot}. Supported: 'teb_12dof', 'new_teb', 'real_teb', 'g1_12dof'"
        )

def _default_base_init(robot: str):
    if robot == "g1_12dof":
        return (2.0, 0.0, 2.5), (0.0, 0.0, 0.5), (0.0, 0.0, 0.78)
    else:
        return (2.0, 0.0, 2.5), (0.0, 0.0, 0.5), (0.0, 0.0, 0.85)

def run(num_envs: int, robot: str, render: bool) -> None:
    gs.init(logging_level="info")

    # Match main.py defaults
    dt = 0.02
    substeps = 4
    camera_pos, camera_lookat, base_pos = _default_base_init(robot)
    base_quat = (1.0, 0.0, 0.0, 0.0)

    n_rendered_envs = num_envs if num_envs < 10 else 5

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(dt=dt, substeps=substeps),
        viewer_options=gs.options.ViewerOptions(
            max_FPS=int(0.5 / dt),
            camera_pos=camera_pos,
            camera_lookat=camera_lookat,
            camera_fov=40,
        ),
        vis_options=gs.options.VisOptions(n_rendered_envs=n_rendered_envs),
        rigid_options=gs.options.RigidOptions(
            dt=dt,
            constraint_solver=gs.constraint_solver.Newton,
            enable_collision=True,
            enable_joint_limit=True,
            batch_dofs_info=True,
            batch_links_info=True,
            enable_self_collision=False,
        ),
        show_viewer=render,
    )

    # Ground plane
    scene.add_entity(gs.morphs.Plane())

    # Robot
    model_path = _select_model_path(robot)
    if model_path.endswith(".xml"):
        robot_entity = scene.add_entity(gs.morphs.MJCF(file=model_path))
    elif model_path.endswith(".urdf"):
        robot_entity = scene.add_entity(
            gs.morphs.URDF(file=model_path, pos=base_pos, quat=base_quat)
        )
    else:
        raise ValueError(f"Unsupported model file: {model_path}")

    # Build and step
    scene.build(n_envs=num_envs, env_spacing=(0.1, 0.1))

    print("Initialized. Stepping physics without actions (no control updates).")
    try:
        while True:
            scene.step()
    except KeyboardInterrupt:
        print("\nStopped.")

def main() -> None:
    parser = argparse.ArgumentParser()
    parser.add_argument(
        "--robot",
        type=str,
        default="g1_12dof",
        help="Robot type (teb_12dof | new_teb | real_teb | g1_12dof)",
    )
    parser.add_argument(
        "-B", "--num_envs", type=int, default=3, help="Number of environments"
    )
    parser.add_argument(
        "--render", action="store_true", help="Enable rendering in the viewer"
    )
    args = parser.parse_args()

    run(num_envs=args.num_envs, robot=args.robot, render=args.render)

if __name__ == "__main__":
    main()
'''

10K envs works fine 
20K envs does not 

--

Can you tell me if it works for you? Could also be an issue with 50 series of GPUs ?

ax-anoop, 2025-09-18:
FYI: 

[image]

[image]

ax-anoop, 2025-09-19:
I am on a 5090, are you able to test on a 50 series card ?

YilingQiao, 2025-09-19:
got it, I can find a 5090 machine to test it

duburcqa, 2025-09-19:
Yes it is the same. Building a team is not an easy job, and Genesis simulator is not a commercial product.

ax-anoop, 2025-09-19:
Well I really hope alot is invested into it, we would definatley pay for Genesis ! It's an awesome simulator. All the bugs have made it a little difficult however, so our primairy sim is Issac, which I hope changes soon. 

Very off topic but this is us: https://finance.yahoo.com/news/waterloo-startup-axibo-ai-secures-130000536.html

Would be great to connect with the team at some point, we've made alot of strides on the hardware side.

YilingQiao, 2025-09-20:
Can you try installing the latest nvidia engine? (this bug should be fixed after 580.76)
for example
https://www.nvidia.com/en-us/drivers/details/254126/

YilingQiao, 2025-09-22:
@ax-anoop can you try the latest nvidia driver? The bug happens when PTX code is JIT complied into binary. The latest nvidia dirver fixed it for 5090.

YilingQiao, 2025-09-29:
This error should be fixed with new nvidia driver. Please reopen this issue if it doesn't work.

closed 2025-09-29 by YilingQiao

ax-anoop, 2025-09-29:
testing right now