Genesis-Embodied-AI/genesis-world #1749: [Feature]: Clear camera recording buffer

jgillick, 2025-09-18:
### What feature or enhancement are you proposing?

It would be nice to be able to clear the camera recording buffer, so you can dispose of the current recorded images and start a new recording. 

This would give training programs the ability to have a temporary buffer that could either be saved, or recorded over, depending on the circumstances. 

### Motivation

This came from a [video environment wrapper](https://github.com/jgillick/genesis-forge/blob/main/genesis_forge/wrappers/video.py) I'm building. This wrapper regularly records videos during training and has the ability to record the final episode when the environment closes. In order to do this, it needs to record the beginning of each episode, and if a new episode begins before `env.close()` has been called, it needs to clear the previous recording and start a new one.

### Potential Benefit

Environments can have a background buffer that can be saved or recorded over.

### What is the expected outcome of the implementation work?

- [ ] At any given point, calling `camera.clear()` will clear the recording buffer.

akushonkamen, 2026-10-18:
Picking this up. Since frames are streamed to the encoder as they are rendered (genesis/vis/camera.py:686-729), 'clearing the buffer' means discarding the recording in progress: close the encoders, delete the partial file(s), and reset the recording state so `start_recording()` opens a fresh video with any filename and framerate. Will add `Camera.clear()` plus a regression test in tests/rendering/test_offscreen.py. (AI-generated fix, developed with an coding agent.)

akushonkamen, 2026-10-18:
Fix submitted as #3566 (adds `Camera.clear()` that discards the recording in progress — closes the encoders, deletes the written video file(s), resets state; no-op when nothing to discard). Includes a regression test (`test_video_recording_clear`); red→green verified.

akushonkamen, 2026-10-18:
Update: review findings addressed in #3566 (972325d) — clear() no longer deletes a completed video saved earlier under the same filename when the new recording got no frames, and cleanup is robust to encoder close() failures.