Genesis-Embodied-AI/genesis-world #1771: [Bug]: Hybrid robot base_link remains fixed even with `fixed=False`

shng1, 2025-09-24:
### Bug Description

When using a hybrid robot under gravity, the robot’s base_link is always fixed to the world coordinate, even when `fixed=False` is specified in the `gs.morphs.URDF()`. As a result, the robot cannot move or fall under gravity.

### Steps to Reproduce

```python
import numpy as np
import genesis as gs

########################## init ##########################
gs.init(seed=0, precision='32', logging_level='debug')

######################## create a scene ##########################
dt = 3e-3
scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        substeps=10,
    ),
    viewer_options= gs.options.ViewerOptions(
        camera_pos=(1.5, 1.3, 0.5),
        camera_lookat=(0.0, 0.0, 0.0),
        camera_fov=40,
    ),
    rigid_options=gs.options.RigidOptions(
        dt=dt,
        gravity=(0, 0, 9.8),
        enable_collision=True,
        enable_self_collision=False,
        contact_resolve_time=0.02,
    ),
    mpm_options=gs.options.MPMOptions(
        dt=dt,
        lower_bound=( 0.0,  0.0, -0.2),
        upper_bound=( 1.0,  1.0,  1.0),
        gravity=(0, 0, 9.8),
        enable_CPIC=True,
    ),
    vis_options=gs.options.VisOptions(
        show_world_frame=True,
        visualize_mpm_boundary=False,
    ),
)

########################## entities ##########################
scene.add_entity(morph=gs.morphs.Plane())

robot = scene.add_entity(
    morph=gs.morphs.URDF(
        file="urdf/simple/two_link_arm.urdf",
        pos=(0.5, 0.5, 0.3),
        euler=(0.0, 0.0, 0.0),
        scale=0.2,
        fixed=False,  # <-- explicitly set fixed to False.
    ),
    material=gs.materials.Hybrid(
        mat_rigid=gs.materials.Rigid(
        ),
        mat_soft=gs.materials.MPM.Muscle( # to allow setting group
            E=1e4,
            nu=0.45,
            rho=1000.,
            model='neohooken',
        ),
        thickness=0.05,
        damping=1000.,
        func_instantiate_rigid_from_soft=None,
        func_instantiate_soft_from_rigid=None,
        func_instantiate_rigid_soft_association=None,
    ),
)

ball = scene.add_entity(
    morph=gs.morphs.Sphere(
        pos=(0.8, 0.6, 0.1),
        radius=0.1,
    ),
    material=gs.materials.Rigid(rho=1000, friction=0.5),
)

base_link = robot.part_rigid.get_link("base")

########################## build ##########################
scene.build()

########################## run ##########################
scene.reset()

for i in range(300):
    dofs_ctrl = np.array([
        1. * np.sin(2 * np.pi * i * 0.001),
    ] * robot.n_dofs)

    robot.control_dofs_velocity(dofs_ctrl)

    print(f"base_link pos: {base_link.get_pos()}, vel: {base_link.get_vel()}")

    scene.step()

```
Make sure to attached any needed assets (here `attachement.xml`)!

### Expected Behavior

With `fixed=False`, the base_link should not be pinned to the origin. The hybridrobot should fall under gravity and move freely, similar to when using a rigid-only URDF like below.

https://github.com/user-attachments/assets/f2dcbde0-ed53-4ac9-97c2-670b7341060d

### Screenshots/Videos

- OS: Ubuntu 20.04
- GPU: RTX 4090
- CPU: Intel Core i9-13900KF
- GPU driver: 575.57.08
- CUDA toolkit: 12.9
- Genesis: 0.3.3

### Relevant log output

```shell
[Genesis] [17:50:50] [INFO] Compiling simulation kernels...
[Genesis] [17:51:09] [INFO] Building visualizer...
[Genesis] [17:51:10] [DEBUG] Trying to create EGL Context for EGL_DEVICE_ID='0'...
[Genesis] [17:51:10] [DEBUG] Resetting Scene <1d7f770>.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:10] [INFO] Running at 10.22 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:10] [INFO] Running at 10.25 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:10] [INFO] Running at 10.25 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:10] [INFO] Running at 10.29 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.34 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.31 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.37 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.39 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.42 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.43 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.43 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.44 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.43 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
[Genesis] [17:51:11] [INFO] Running at 10.47 FPS.
base_link pos: tensor([0.5000, 0.5000, 0.3000]), vel: tensor([0., 0., 0.])
```

### Environment

- OS: Ubuntu 20.04
- GPU: RTX 4090
- CPU: Intel Core i9-13900KF
- GPU driver: 575.57.08
- CUDA toolkit: 12.9

### Release version or Commit ID

- Version number: 0.3.3 (and all 0.3.x and 0.2.1)

### Additional Context

I opened another issue of hybrid body instability: https://github.com/Genesis-Embodied-AI/Genesis/issues/1769

yuanchang20021217-netizen, 2025-09-28:
I am an amdgpu user and I can replicate this bug on vulkan backend as well. After perform some test I doubt it`s a gstaichi issue. This is my test: I can recreate this bug on 0.3.x and 0.2.1. And I uninstalled gstaichi and use 0.2.1 and it`s working fine!

shng1, 2025-10-17:
I confirmed that the issue still reproduces with the latest commit: https://github.com/Genesis-Embodied-AI/Genesis/commit/c413fdeae45e8192955d8c35e90474e7459e263a
I've also updated the reproduction code to reflect the recent API parameter changes.